fieldOfView/Cura-LinearAdvanceSettingPlugin #3: Custom printer definition causes Cura to crash when this plugin is installed

johnkray, 2018-11-15:
Hi,

This issue could be on my end, but I thought I'd reach out and see if you have any idea what might be causing this. 

I'm working on Cura configuration for a printer I am developing. Everything works fine with the definition I made, but when I try to install the Linear Advance Plugin, or really any plugin that adds settings fields to the sidebar, it causes Cura to crash. 

I've tested this several time, the plugin and my definition work fine on their own, but when both are installed Cura won't launch. I've tested this on both MasOS and Window and the problem persists on both.

Any ideas?

fieldOfView, 2018-11-15:
There is nothing I can do about this without your custom printer definition, or logs, or both.

johnkray, 2018-11-15:
Here is the definition. If you need to see the variants, quality, extruders etc, let me know.

    "version": 2,
    "name": "Hydra Research Nautilus",
    "inherits": "fdmprinter",
    "metadata": {
        "visible": true,
        "author": "Hydra Research",
        "manufacturer": "Hydra Research",
        "file_formats": "text/x-gcode",
        "icon": "icon_ultimaker2",
        "platform": "hydra_research_nautilus_platform.stl",
		"has_machine_quality": true,
        "has_materials": true,
        "has_machine_materials": true,
        "has_variant_materials": true,
		"has_variants": true,
        "variants_name": "Tool Cartridge",
        "preferred_variant_name": "BR 0.40",
		"preferred_quality_type": "Fast",
        "machine_extruder_trains":
        {
                        "0": "hydra_research_nautilus_extruder_0",
			"1": "hydra_research_nautilus_extruder_1",
			"2": "hydra_research_nautilus_extruder_2"
			
        }
    },

    "overrides": {
        "machine_name": { "default_value": "Hydra Research Nautilus" },
        "machine_heated_bed": { "default_value": true },
        "machine_width": { "default_value": 205 },
        "machine_height": { "default_value": 225 },
        "machine_depth": { "default_value": 205 },
        "machine_center_is_zero": { "default_value": false },
        "material_bed_temperature": { "default_value": 50 },
        "retraction_amount": { "default_value": 0.8 },
        "retraction_speed": { "default_value": 35 },
        "retraction_retract_speed": { "default_value": 35 },
        "retraction_prime_speed": { "default_value": 35 },
        "adhesion_type": { "default_value": "none" },
        "machine_head_with_fans_polygon": { "default_value": [[-31,31],[34,31],[34,-40],[-31,-40]] },
        "gantry_height": { "default_value": 28 },
		"machine_max_feedrate_x": { "default_value": 600 },
        "machine_max_feedrate_y": { "default_value": 600 },
        "machine_max_feedrate_z": { "default_value": 40},
        "machine_max_feedrate_e": { "default_value": 120 },
        "machine_max_acceleration_z": { "default_value": 1200 },
        "machine_acceleration": { "default_value": 1200 },
        "machine_max_jerk_xy": { "default_value": 8 },
        "machine_max_jerk_z": { "default_value": 0.2 },
        "machine_max_jerk_e": { "default_value": 2.5 },
        "machine_gcode_flavor": { "default_value": "RepRap (RepRap)" },
		"machine_extruder_count": { "default_value": 3 },
        "machine_start_gcode": {
            "default_value": "M98 P/sys/messageprintstart.g\nM104 S{material_print_temperature_layer_0} T0 ; set extruder temp\nM140 S{material_bed_temperature_layer_0} ; set bed temp\nG32 ; home xy\nM190 S{material_bed_temperature_layer_0} ; wait for bed temp\nM109 S{material_print_temperature_layer_0} T0; wait for extruder temp\nM98 P/sys/homeu.g\nG1 X-12.5 Y0 F6000 ; Position Tool Head for filament change\nM83\nG1 U0 F2000 ; Position filament selector for bay 1\nM98 P/sys/filamentload{max_feedrate_z_override}.g\nM98 P/sys/filamentprime{max_feedrate_z_override}.g\nM83\nM116"
        },
        "machine_end_gcode": {
            "default_value": "M98 P/sys/printend.g" 
			},
		

		
		"layer_height_0": { "value": "round(machine_nozzle_size / 1.6, 2)" },
		"line_width": { "value": "round(machine_nozzle_size * 1.125, 3)" },
		"wall_line_width_x": { "value": "round(wall_line_width, 2)" },
		"infill_line_width": { "value": "round(line_width, 2)" },
		"roofing_line_width": { "value": "machine_nozzle_size" },
		"initial_layer_line_width_factor": { "value": "89" },
[... 38 lines cut ...]
		"retraction_hop_only_when_collides": { "value": "True" },
		"retraction_hop": { "value": "0.6" },
		"retraction_hop_after_extruder_switch": { "value": "True" },
		
		"cool_fan_full_at_height": { "value": "layer_height_0 + 4 * layer_height" },
        "cool_fan_speed": { "value": "50" },
        "cool_fan_speed_max": { "value": "100" },
        "cool_min_speed": { "value": "20" },
	
		"support_angle": { "value": "50" },
        "support_pattern": { "value": "'zigzag'" },
        "support_use_towers": { "value": "False" },
        "support_xy_distance": { "value": "wall_line_width_0 * 2.5" },
        "support_xy_distance_overhang": { "value": "wall_line_width_0" },
		"support_interface_enable": { "value": "True" },
		"support_interface_height": { "value": "1" },
		
		"slicing_tolorance": { "value": "'middle'" },
		"coasting_enable": { "value": "True" },
		"bridge_settings_enabled": { "value": "True" },
		"bridge_enable_more_layers": { "value": "False" }
   
		
    }
}

johnkray, 2018-11-15:
Here is the error log

Faulting application name: Cura.exe, version: 0.0.0.0, time stamp: 0x5bd2dc78
Faulting module name: ucrtbase.dll, version: 10.0.17134.319, time stamp: 0x40b70dec
Exception code: 0xc0000409
Fault offset: 0x000000000006e57e
Faulting process id: 0x42a8
Faulting application start time: 0x01d47d312c0a7762
Faulting application path: C:\Program Files\Ultimaker Cura 3.6\Cura.exe
Faulting module path: C:\WINDOWS\System32\ucrtbase.dll
Report Id: 2f222933-97ea-4212-9ef3-47494a4ded23
Faulting package full name: 
Faulting package-relative application ID:

fieldOfView, 2018-11-16:
I would need the printer definition, extruder(s) and variants. Please zip them up, don't paste them in here.
The log I was looking for is cura.log from your configuration folder.

fieldOfView, 2018-11-18:
The crash also happens with other plugins (Printer Settings, Z Offset, and others based on these). The plugins are not at fault here, your custom definitions are.

There are two issues with your custom definitions and profiles, which cause an issue:
* The mh_build_abs.xml.fdm_material file references a variant that does not exist (HS 0.25). This should be removed (or the variant should be added)
* The quality profiles all use the wrong setting_version. For Cura 3.6, setting_version must be set to 5; it is not the version of the file (for your bookkeeping), but the version of the settings, used to determine if the files need to be upgraded).

After making these changes, the definitions and profiles work in combination with the plugin.

closed 2018-11-18 by fieldOfView