fieldOfView/Cura-LinearAdvanceSettingPlugin #5: Installation Prevents Saving 3MF Files on MacOS

kazooless, 2019-02-21:
Running MacOS 10.14, immediately after installing this plugin from the Marketplace, Cura will no longer save 3MF files for me. If I uninstall it, Cura can save again.

fieldOfView, 2019-02-22:
> Cura will no longer save 3MF files for me.

Please elaborate. What happens exactly?
Can you please share your cura.log (from Help->Show configuration folder)

kazooless, 2019-02-22:
I've narrowed it down a little more. If I set Linear Advance Factor to zero, then it saves. If there is a setting in there, then literally nothing happens when I hit the save button. The same for trying with "Export..." instead. So, Go to File -> Save, choose the location and the name, click "Save," and there is no dialog and not file has been saved. Gcode saving is fine.

[cura.log](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/files/2895708/cura.log)

fieldOfView, 2019-02-23:
Thanks, I have spotted my mistake.

closed 2019-02-23 by fieldOfView

fieldOfView, 2019-02-23:
The problem is in saving twice. You first export gcode, then save a 3mf file. If you were to first save the 3mf file and then try to export gcode, writing the gcode would fail. If you would reslice inbetween, it would work. 

This (silly) bug has been fixed, I have submitted a new version to the Marketplace. It will take a little while for it to show up there. In the mean time you can use this patched version:
http://files.fieldofview.com/cura/LinearAdvanceSettingPlugin-v5.0.0-2019-02-23T08_24_54Z.curapackage
Download the file and drop it into the Cura application window.

kazooless, 2019-02-23:
Awesome. Glad it was easy for you. 

Thank you,

_____
Jeff Kazules

fieldOfView, 2019-02-26:
Fixed version is now in the Marketplace.

commit 420b20e4 (2019-02-23): Fix missing import

fixes #5