fieldOfView/Cura-LinearAdvanceSettingPlugin #6: Feature Request: Add LA Factor to Material Settings

DruckiMcDruck, 2019-04-08:
Hi,
First and foremost: thanks for this plugin and your work on Cura itself. Much appreciated. Now, my suggestion:

Currently you add the LA Factor to the Custom print profile. However with Cura 4.0 you can configure different materials and brands of filament (like PLA, PETG, ABS) and use those material profiles to for example adjust Hotend and Bed temperature. So you basically keep your print profile, change the material and voila, you got the correct temps for the chosen material. 

I'd love to see this for LinearAdvance too. So instead of having to maintain different print profiles for various LA values, you'd just like the LA value to the material database. This way you can still use a single custom print profile, but adjust LA by just switching materials. 
[image]

DruckiMcDruck, 2019-04-10:
Just noticed that this was asked before... Must've overlooked the older issue when i created this. Sorry for that.

closed 2019-04-10 by DruckiMcDruck

fieldOfView, 2019-04-28:
I have created a plugin that lets the user select which print settings are available on the Materials pane of the preferences. It is still a bit rough, but an initial version can be downloaded here:

Cura 3.5/3.6: http://files.fieldofview.com/cura/MaterialSettingsPlugin-v5.0.0-2019-04-27T19_48_03Z.curapackage
Cura 4.0: http://files.fieldofview.com/cura/MaterialSettingsPlugin-v6.0.0-2019-04-27T19_48_03Z.curapackage

After downloading, just drop the file into a running Cura window, and restart Cura when requested. By default, the plugin adds the Flow setting to the Print Settings tab of materials on the Materials pane of the preferences.

The plugin also adds a dialog to configure additional settings to be included in the material Print Settings, see Extensions -> Material Settings -> Configure Material Settings. Using that dialog it is possible to add other settings such as Z Offset or Linear Advance Factor if you have the respective plugins for that installed.

Feedback is appreciated.

DruckiMcDruck, 2019-04-28:
I just tested this on Cura 3.6.0 and it seems to work. I can add Curas standard settings and even custom settings (eg LA plugin) to the materials. However it seems that i can only assign integer values to custom settings. So eg. setting the LA factor to 1 or 2 works, but 0,5 doesn't. Setting for example the line width to 0,23mm does also work. Thus my conclusion is that this is only an issue for custom values added by plugins. 

I'm running a modified version of your LA plugin where i added some additional settings. This issue happens on both, my modified version and your LA plugin: 

[image]

Thanks for making this additional plugin. It'll save me from a lot of messing around with different printer profiles for various materials.

fieldOfView, 2019-04-28:
Thanks for testing.

This was a weird bug to trace down, but it turns out there is some very crufty code in the Materials preference pane that checks if the *unit* of a setting is `mm`:
https://github.com/Ultimaker/Cura/blob/4.0/resources/qml/Preferences/Materials/MaterialsView.qml#L441
This will need to be fixed in Cura unfortunately. In the mean time you can lie to Cura from your plugin and make the unit `mm`; the unit is only cosmetic anyway.

DruckiMcDruck, 2019-04-28:
Thanks for the quick response. I might take that route of adding the `mm` as unit, if it fixes the issue as it probably won't get fixed for Cura 3.x anymore and i'm not ready to migrate to Cura 4.x yet. Also thanks for investigating and pointing out the issue. 

This is the repo of the plugin you linked above i assume: https://github.com/fieldOfView/Cura-MaterialSettingsPlugin If i encounter another issue with it, i'll report it over there. 

Thanks for your kind support and the new plugin! :)

fieldOfView, 2019-06-14:
I have found a way to patch this up from inside the plugin. A testing version is available here:
http://files.fieldofview.com/cura/MaterialSettingsPlugin-v6.0.0-2019-06-14T14_49_58Z.curapackage