fieldOfView/Cura-LinearAdvanceSettingPlugin #9: Can't disable linear advance (set 0) in 3.5.5

MatkovIvan, 2019-07-02:
`0` is valid value, but isn't allowed anymore.
[image]
[image]

Please fix minimal value:
https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/blob/1cc2732d308d84df2ef846bc5acef79d1232bedc/LinearAdvanceSettingPlugin.py#L27

fieldOfView, 2019-07-02:
Sorry, my thought was you would set the linear advance factor to 1 to disable it. It seems I was mistaken. I'll submit a fixed version of the plugin.

closed 2019-07-02 by fieldOfView

fieldOfView, 2019-07-12:
The fixed version is now available in the Marketplace.

MatkovIvan, 2019-07-12:
Yeah, thanks.

Fortunately, it's distributed as sources, so I was able to fix it locally.

commit ed5156d3 (2019-07-02): Allow setting linear advance factor to 0

Fixes #9