fieldOfView/Cura-LinearAdvanceSettingPlugin #10: Default to 0 instead of 1?

randellhodges, 2019-10-18:
In Marlin 2 with linear advance version 1.5, a value of 0 is supposed to be "disabled". With most settings seeming to be in the 0.09 to 0.11 range, a default value of 1 might be a little high.

Also, in the readme, I see you mention M900 with other settings, but looking at the marlin documentation for M900 I don't see those values. Maybe that is for an older version of linear advance?

fieldOfView, 2019-10-18:
From the documentation:
> While old v1 K values for PLA might be between 30-130, you can now expect K to be around 0.1-2.0

The value is described as a factor, and normally using a factor of 1 implies no change. But I get your point, and will change the default in the next version of the plugin.

fieldOfView, 2019-10-18:
How would you change the README?

closed 2019-10-18 by fieldOfView

randellhodges, 2019-10-18:
> How would you change the README?

M900 W{line_width} H{layer_height} D{material_diameter}

http://marlinfw.org/docs/gcode/M900.html

I don't see (anymore?) a reference to W/H/D. I didn't look at the code, just the documentation.

commit 72f27565 (2019-10-18): Set default value to 0 ("off")

Fixes #10