fieldOfView/Cura-LinearAdvanceSettingPlugin #12: Multiple M900 causing pauses

randellhodges, 2019-11-24:
I seem to have a ton of these now being inserted:

;TYPE:XXX
M900 K0.090000 T0 ;added by LinearAdvanceSettingPlugin

When Marlin hits one of these, it causes the print head to pause which is causing defects in the print.

I only have a single extruder, so I think 1 would be enough. Maybe keep track of the last `extruder_nr` and only insert the M900 if it changes. I'll take the tweak on my side and see if that helps.

fieldOfView, 2019-11-24:
I don't know why Marlin would pause on an M900 command. Then again, I don't have any printers that support Linear Advance, so I don't use the plugin myself; I rely on reports like yours to know if things work or not. So thanks for the report.

I'll implement a fix like yours, but taking into account the feature type as well.

As an experiment, I would be interested to know if the pause also happens if you remove the T0 argument (given that you have only a single extruder).

randellhodges, 2019-11-24:
Here is snippet from Marlin:

```
    if (parser.seenval('K')) {
      const float newK = parser.value_float();
      if (WITHIN(newK, 0, 10)) {
        planner.synchronize();
        planner.extruder_advance_K[tool_index] = newK;
      }
      else
        SERIAL_ECHOLNPGM("?K value out of range (0-10).");
    }
```

```
/**
 * Block until all buffered steps are executed / cleaned
 */
void Planner::synchronize() {
```
Basically, the planner stops ingesting commands. I had mine set to 32 and ADVANCED_OK enabled. I was watching Octoprint and saw at one of these M900s, the P in advanced ok, go from 0 (meaning a full planning buffer) up to 32, then a slight pause in Octoprint with a [...] printed, and the print heat pause, just for a second, then everything go.

My test object was a 30x30x10 cube with rounded corners. The rounded corners probably made the problem worse. I almost dropped Cura and went to PrusaSlicer because it didn't have issues. I just thought it was Cura slicing too many small segments on the corners.

I don't fully understand the Marlin/Octoprint serial communication with all the buffers it has. I wouldn't really expect this to be an issue and actually feel it is something that would be better addressed in Marlin. I feel these types of actions should be queued inside the planner buffer and just executed as another step and not sync/lock the buffer before making the change.  Sort of just push a lambda onto a stack sort of approach.

I do appreciate the plugins and the effort you put into them. They are probably the largest reason I continue to use Cura.

fieldOfView, 2019-11-24:
Could you try this version?
https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/commit/502bbafde899a4ef836d113c7e0cea9537e6d6ae

This puts back the M900 command right after the start gcode (like in previous versions), and only adds additional M900 commands for feature-types if they differ from the "main" setting. It is mostly your code, with an optimization to check if the full gcode needs to be processed at all.

randellhodges, 2019-11-24:
> As an experiment, I would be interested to know if the pause also happens if you remove the T0 argument (given that you have only a single extruder).

With a single extruder, Marlin doesn't even look at the T command. It isn't even compiled in:

  ```
#if EXTRUDERS < 2
    constexpr uint8_t tool_index = 0;
  #else
    const uint8_t tool_index = parser.intval('T', active_extruder);
    if (tool_index >= EXTRUDERS) {
      SERIAL_ECHOLNPGM("?T value out of range.");
      return;
    }
  #endif
```

fieldOfView, 2019-11-24:
> I don't fully understand the Marlin/Octoprint serial communication with all the buffers it has. 

Who does?
https://media.tenor.com/images/ef67c19e7ad139ea0fe35381dd1c1c92/tenor.gif

randellhodges, 2019-11-24:
> Could you try this version?
> [502bbaf](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/commit/502bbafde899a4ef836d113c7e0cea9537e6d6ae)
> 
> This puts back the M900 command right after the start gcode (like in previous versions), and only adds additional M900 commands for feature-types if they differ from the "main" setting. It is mostly your code, with an optimization to check if the full gcode needs to be processed at all.

Getting an error slicing:

```
Exception: Unable to write to file single_wall_box: 'LinearAdvanceSettingPlugin' object has no attribute '_setting_key'

2019-11-24 14:57:15,848 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\...\AppData\Roaming\cura\4.4\plugins\LinearAdvanceSettingPlugin\LinearAdvanceSettingPlugin\LinearAdvanceSettingPlugin.py", line 128, in _filterGcode
2019-11-24 14:57:15,850 - ERROR - [MainThread] UM.Logger.logException [88]:     linear_advance_factor = extruder_stack.getProperty(self._setting_key, "value")
```

fieldOfView, 2019-11-24:
...which is why I should not be doing things on a sunday evening ;-)

Here's a new one (which I have actually tested a bit): 9500d878bf3817aa407673aaefecc1d3ba35f32f

randellhodges, 2019-11-24:
> ...which is why I should not be doing things on a sunday evening ;-)
> 
> Here's a new one (which I have actually tested a bit): [9500d87](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/commit/9500d878bf3817aa407673aaefecc1d3ba35f32f)

Thanks. I can confirm I can generate gcode now and that I only see it once. I don't use all the suboptions so I won't be able to verify quickly.

One little thing, if it isn't too much trouble. Can the first instance of M900 be inserted after `;LAYER:0`? I'm a little fussy with things before my custom machine gcode. It certainly won't hurt to have it "up there" but after would make my OCD happy :)  Either way, thanks for looking into it.

fieldOfView, 2019-11-24:
Thanks for testing.

> One little thing, 
Done

closed 2019-11-24 by fieldOfView