fieldOfView/Cura-LinearAdvanceSettingPlugin #13: Fork Request: G29 Mesh Slot Plugin

Careless, 2019-12-09:
Hi,

I use your linear advance plugin and it is very useful for setting K constant per settings profile. Is it possible to create a similar plugin to load a G29 Mesh slot (as per marlin Gcode) that would select the appropriate mesh save slot that a particular map is saved within for use with a build platform that is selectively used to print certain materials?

Essentially like this plugin, it would have "Mesh Slot" with an input field for a 1 digit positive/integer number which would circumvent having to put the G29 code in the start g-code.

It would be great to have it show up right under the linear advance input field from this plugin.

fieldOfView, 2019-12-11:
Would you want to be able to change this setting per material? How should this behave for multiple extruders?

Careless, 2019-12-11:
Hmmm.. I think material would be best, yes. Different materials require different build platforms, so that would be the most practical.

What do you think? If someone is printing nylon, they might switch out to a phenolic sheet, and if someone switches to PLA they might print to buildtak or PEI or straight to glass- so material seems to be ideal for the mesh slot setting.