fieldOfView/Cura-LinearAdvanceSettingPlugin #18: Apparently a crash?

Ghostkeeper, 2020-04-23:
Hey there. I saw a crash go by through Cura's issue reporter that occurs in the code of this plug-in:
https://sentry.io/share/issue/ff9a2d954c384a52ab9652e8e182e097/

I'm not entirely sure what to do with this. It's not able to find a container called `artillery_sidewinder_x1_0.4`, but apparently you are getting an `_onContainerLoadComplete` call from it. This Artillery Sidewinder is not a printer that's by default in Cura, and I know it as a printer that people regularly mod into Cura by copy-pasting files in there.

I'm unable to reproduce the crash myself. Maybe this could happen somewhere in multi-threading, if something deletes the variant before it gets to process your callback? It looks like this is a nozzle profile so you wouldn't be interested in it anyway. In any case, maybe it'd be good to skip this function if the ID doesn't exist.

fieldOfView, 2020-04-23:
I think this will have already been fixed with this commit (which is not yet in the released version): https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/commit/adfa12c35e56bb1e722563adc7e21586c50ce84c
But I'll add some more defensive programming around there. Thanks for the headsup.

Ghostkeeper, 2020-04-23:
We're also looking into why it's calling _onContainerLoadComplete with an ID that doesn't exist, because more plug-ins seem to have this issue. Always with these Artillery printers though.

fieldOfView, 2020-04-23:
Looking into the crash report more I can confirm that it is the issue I fixed with adfa12c.

What happens is that when a container can not be deserialised (because eg its definition is missing, or its settingversion is too new, or things like that), the ContainerLoadComplete signal is still emitted. The container id actually exists in the ContainerRegistry, and doing a ContainerRegistry.findContainers() then tries to load the container. This fails, but the ContainerLoadComplete signal is still emitted. My module gets the newly (not) loaded container id, and does a ContainerRegistry.findContainers() with it, which tries to load the container, which fails. Well, you get the infinitely recursing point.

The fault lies with me for not checking if the container is loaded before trying to find it. It is also with Cura (or Uranium) for saying the load completed, when it didn't.

PrinterSettingsPlugin and ZOffsetPlugin have the same issue, and have the same fix waiting to be released.

Also see https://github.com/Ultimaker/Cura/issues/7393

closed 2020-05-31 by fieldOfView