fieldOfView/Cura-LinearAdvanceSettingPlugin #19: Pressure Advance Setting for Klipper Firmware

Henry3DP, 2020-05-31:
Hi,

I am new to all this, so I hope this is the right place to ask. 

Would it be possible to extend the plugin so that you can also set the pressure advance for Klipper Firmware?

fieldOfView, 2020-05-31:
Is pressure advance for Klipper different from pressure advance for reprap firmware?

fieldOfView, 2020-05-31:
Referring to https://github.com/KevinOConnor/klipper/issues/293#issuecomment-381717654, would adding eg `SET_PRESSURE_ADVANCE ADVANCE=0.4` to the gcode instead of `M90 K0.4` be enough for you, or would you need more?

Henry3DP, 2020-05-31:
Thats exactly what I am looking for!

fieldOfView, 2020-05-31:
Is there any way Cura could know you are using Klipper? Currently the LinearAdvance plugin checks the "firmware flavor" to decide between adding Linear Advance for Marlin, and Pressure Advance for Reprap firmware. But I think that for Klipper, you also slice with Marlin flavored gcode, right?

Henry3DP, 2020-05-31:
Yes unfortunately for Klipper you also use Marlin flavoured gcode. 

If this makes the integration of this function impossible. Is there maybe a way to change it as a user? I mean as someone who has no idea about programming and would just follow instructions.

fieldOfView, 2020-07-02:
I have added a setting named "Advance Flavor" that lets you select Klipper pressure advance. I have not given it more than a "smoke test" to see if it inserts something resembling what you needed, so it would be cool if you could test it. Download this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/LinearAdvanceSettingPlugin/3t-MEFEJzhSwLGnoz9Q1rKjE64-z4P7qn-OF88nnrXXx/LinearAdvanceSettingPlugin-v7.0.0-2020-07-02T14%3A10%3A15Z.curapackage?Expires=1909059018&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=Dw%2FlGdbmGyGynSuvZqDyNhSAX1C%2Bogt%2B798RZqFvJfECrF1tAwj5EEQl8iAJD66tYTcPnztN5TMvrNGy8hRWBPzLOU%2BrdpADFVxjvfIxWltgpgHNmYtfXYgJYSoFyhMy3tE%2BO7Hw4BoUzoze3gTmupeiFP2IiCb0PQHeQy7Z6%2BA8GoeT%2Fwf0DMgaWXKS7RERhfKJ%2Brv2LJQ2tmsS5bgnio49%2F3ZDFGdpykOmLSX%2FtKHmbRyT5c4jfy4hu4P%2FgMtmM4DrMJ%2B3eHtJvL4b5za1zHqi6jzioYtn%2B9%2F6nCDaZWc%2FME0YGIThyovTNovdU2Ov7jnotGz7rWifixd97z6Wxw%3D%3D) and drop it onto the Cura buildplate as if you were opening a 3d model.

Asterchades, 2020-08-14:
This can be done Klipper-side without too much hassle. I'm honestly not 100% sure why Mr O'Connor hasn't opted to use the standard Marlin command given the functionality is essentially interchangeable between the two features, but that's neither here nor there really.

Just put the following in printer.cfg (I don't think it matters where, but I have it at the bottom):

[gcode_macro m900]
default_parameter_K: 0
gcode:
  SET_PRESSURE_ADVANCE ADVANCE={K}

fieldOfView, 2020-09-04:
I have decided to revert the changes to directly support Klipper, and instead added the gcode macro to the README.

ollyfg, 2020-10-03:
FYI I have a pretty basic plugin to set pressure advance. I haven't added all the options this one has, it just inserts the basic `SET_PRESSURE_ADVANCE ADVANCE={x}` but does the trick for me.

Here: https://github.com/ollyfg/cura_pressure_advance_setting

I'm not really a fan of the m900 macro, because I believe the units are different between pressure advance/ linear advance? I like to keep things explicit if I can, but that's just a personal opinion!

Asterchades, 2020-10-04:
> I'm not really a fan of the m900 macro, because I believe the units are different between pressure advance/ linear advance?
I don't know that they're entirely different units since the introduction of Linear Advance v1.5. LAv1.5 is noted as being a measurement of mm/mm/s (mm of compression per 1mm/s extrusion speed), where Pressure Advance is based on a coefficient being applied to the extrusion speed which is basically the same thing.  The difference, if any, is going to be in their application, but I suspect the units are now the same.

In either case, my experience has been that the correct value for one is extremely close to the correct value for the other on the same hardware and material setup.

At least it's not like Junction Deviation against Square Corner Velocity, which are functionally identical but use wildly different values. I have to say I very much prefer Klipper's implementation there, since it's both more human readable and decoupled from other settings, where JD's effective value will change based on your acceleration.

closed 2020-12-10 by fieldOfView