fieldOfView/Cura-LinearAdvanceSettingPlugin #20: Option to enable at layer N or disable for initial layer

KimmoHop, 2020-06-20:
My first layers do not print well with LA on, and it might not really be that much needed at first layer speed.

It would be nice if this plugin had option to disable LA for initial layer (or have separate settings for initial layer, but that would really be too many settings) or option to enable LA only at layer N or at height H forward.

fieldOfView, 2020-06-23:
I'll add a (single) setting for the linear advance setting to use for the initial layer. If that is set to 0, no linear advance will be applied for the whole initial layer.

fieldOfView, 2020-07-02:
Implemented in 5d84335. I'll leave this issue open until a version that includes this feature is released.

fieldOfView, 2020-07-03:
It would be great if you could help test this feature. Download this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/LinearAdvanceSettingPlugin/3t-MEFEJzhSwLGnoz9Q1rKjE64-z4P7qn-OF88nnrXXx/LinearAdvanceSettingPlugin-v7.0.0-2020-07-02T14%3A10%3A15Z.curapackage?Expires=1909133875&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=O95AoooY9u3EXjfOpKhYLqZ7Vu2A3TRZ8X%2FX7O%2Bz3tI%2FKrdvMKF3nRVLjI0oB5MojOzQ5UuYLnYo4RJDPMR1%2FRf8pkJaQVzZ%2Flkmw%2BM0Ul8yc6BSPwvoeyW5eHI4rt124nj7zkuSVFjUa6%2Bo4sUREQDolhdmWVwR%2FEmFJPJ5H4L%2B7ocR7PXGKp%2FhUetYHLBMZnxprg4Zm%2F76E6k%2FfINcavzDv1kFj6Ha50fgRyMSpIGXu1%2FzAYS48r3%2BRN4RijxLP7wRQfMrPzt%2FSFZJ3yCB1UsAjT322qYiKuOeEZEKUWrI0HXpI8ThYecyugofYmXJB3SU6VHOwK1ff3TTrZ8yZg%3D%3D) and drop it onto the Cura buildplate as if you were opening a 3d model.

KimmoHop, 2020-07-03:
When saving gcode, I get error "name 'material_linear_advance_factor' is not defined". In Cura 4.4 it's only in log file (slice and save parts attached), in 4.6.1 there is also popup (attached)
Unfortunately project could not be saved due to same error, so that is not included.

[image]
[slice_save_log_LA0.5.log](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/files/4870811/slice_save_log_LA0.5.log)

fieldOfView, 2020-07-03:
Thanks, I’ll look into it.

KimmoHop, 2020-08-26:
Actually, after printing more 1st layers, it feels like perimeter order (outer first can be good choice sometimes), small geometries, correct nozzle height / line width (squish) and filament itself make bad 1st layers, linear advance maybe not so much.
So if this issue is problematic and concerns no other users than me, I'm OK with closing it.