fieldOfView/Cura-LinearAdvanceSettingPlugin #21: Add documentation of incompatible cura options.

matiasdelellis, 2020-06-27:
Hi @fieldOfView 

Marlin's documentation is a bit vague about it.

> Some slicers have options to control the nozzle pressure. Common names are: Pressure advance, Coast at end, extra restart length after retract. Disable these options as they will interfere with Linear Advance.
> Also disable options like wipe while retract or combing. There should be almost no ooze, once the proper K-Factor is found.
> Recheck retraction distance, once Linear Advance is calibrated and working well. It may even be as low as 0, since pressure control reduces the material pressure at the end of a line to nearly zero.

Of course they must focus on the firmware and cannot exemplify every slicer.  So I think your plugin can be a good place to properly document this options for Cura.

As you surely have more experience with Linear Advance, I want to ask you what exactly these options are.

This could be documented in the readme, but it would be interesting to show it in the same GUI as a tooltip, warning, or disable it directly.

Thanks for all,
Regards
Matias.

fieldOfView, 2020-06-27:
> As you surely have more experience with Linear Advance, I want to ask you what exactly these options are.

Shall I tell you a "secret"? I have never used this plugin myself. None of my Ultimaker printers support linear advance. I wrote and maintain it because people were requesting it.

If something like this is going to be documented, that documentation will have to come from someone who knows what settings affect linear advance, and preferably also pressure advance for reprap firmware. Pull requests are welcome.

matiasdelellis, 2020-06-27:
haha. Ok..  Let's hope someone who knows more. :see_no_evil:

tonnic, 2021-03-22:
I came to ask the question if the plugin removes those values from the unsupported cura settings, which I am guess it does not now... so we will need to remove any settings in Cura that LA affects, correct?

fieldOfView, 2021-03-22:
The plugin only adds M900 or M572 commands. Other than that it changes nothing in how the model is sliced.