fieldOfView/Cura-LinearAdvanceSettingPlugin #22: Plugin no longer on cura market

wheelyjoe, 2020-08-04:
Is this still active? Or have I missed something? 

Thanks

fieldOfView, 2020-08-04:
As far as I know, the plugin is available in the Marketplace:
https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/LinearAdvanceSettingPlugin

The plugin may not be available in Cura Enterprise, but that is beyond my control.

wheelyjoe, 2020-08-04:
[image]

This is what I see whenever I try and get it. Not an enterprise account.

fieldOfView, 2020-08-05:
Very strange. What is the exact URL of that page?

What if you try from inside Cura?

wheelyjoe, 2020-08-06:
I have the same from inside cura.

https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/LinearAdvanceSettingPlugin
This is the exact URL.

The page works until I log in, once logged in I get the page I screenshot
to you earlier.

fieldOfView, 2020-08-06:
Can you zip and attach the response you get from this url:
https://api.ultimaker.com/cura-packages/v1/cura/v7/packages

You should be able to download the plugin directly from this URL:
https://api.ultimaker.com/cura-packages/v1/cura/v7.0.0/packages/LinearAdvanceSettingPlugin/download
Once downloaded, drag the file onto the buildplate in Cura as if you were opening a 3d model.

fieldOfView, 2020-08-06:
Are you logged in on the Marketplace at all? Could you try logging out?

Ghostkeeper, 2020-08-06:
I have two accounts: One as an "Essentials" user and one normal. When I log into the normal account I can still see the plug-in. When I log into the "Essentials" account I get the same error message as you. This makes sense because the Essentials accounts are more limited to only be able to download plug-ins which have had a 3rd party security review.

I'd expect that if you go to https://marketplace.ultimaker.com/app/cura/plugins you would see 19 plug-ins, right? And 41 plug-ins if you're logged out?

It would be strange if your account somehow got to be an Essentials user, since those are currently only given out to Ultimaker employees as far as I know.

fieldOfView, 2020-08-06:
@wheelyjoe, could you send me your Ultimaker account username or email address, so I can have the Ultimaker Marketplace team investigate? You can send it to aldo (at) fieldofview (dot) com

fieldOfView, 2020-09-03:
From what I have been told, this is an architectural error in the Marketplace backend. Someone signing up for an Essentials account with an email address that has the same domainname as yours will mark you as an Essentials user. I hate that it is happening, but there is nothing I can do about it except complain... loudly. So I did.

For now, if you log out of your Ultimaker account in Cura, you should be able to install all plugins.

closed 2020-09-03 by fieldOfView