fieldOfView/Cura-LinearAdvanceSettingPlugin #26: Error saving files when using dual extruding with K values set

JohnTasto, 2020-12-03:
When attempting to use a separate extruder for support, I get an `Error 1` when saving a GCode file or sending it to OctoPrint, or even when saving a Cura project file once it is sliced.

I'm using Cura 4.8 on Windows. I use Extruder 1 as my main extruder, and it has extensive speed/acc/K adjustments. Extruder 2 is my support extruder and has no explicit K values set. When I remove all K adjustments from Extruder 1 or change the support extruder back to Extruder 1, the files save just fine.

Setting the adhesion extruder to Extruder 2 doesn't seem to cause any issues.

[switch_panel.zip](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/files/5633135/switch_panel.zip)

JohnTasto, 2020-12-03:
I found when setting the same K values for both extruders, GCode files can again be saved. If a single setting differs, however, the error returns.

This is an adequate workaround for now since support doesn't need to be perfect.

fieldOfView, 2021-02-23:
Though your report predates it, I am marking this as a duplicate of #31. It is fixed for the next release of the plugin.

closed 2021-02-23 by fieldOfView