fieldOfView/Cura-LinearAdvanceSettingPlugin #28: Feature request: Allow setting K values per mesh

JohnTasto, 2020-12-03:
The results from the Marlin calibration pattern don't seem to translate to real world prints, so I'd like to be able to calibrate K by printing an array of test objects, each with a different K.

Alternatively, if K could be adjusted in the 'ChangeAtZ' post processing plugin, one could create K towers similar to what Klipper provides for calibrating pressure advance. I've already made K towers for my printer by hand editing G Code, but other Marlin users might be interested in K arrays or K towers.

kdkwarhead, 2020-12-17:
”if K could be adjusted in the 'ChangeAtZ' post processing plugin, one could create K towers”
I was about to do this but just found out that you cant do it with'ChangeAtZ'. I hope this will be added to Cura.

exzile, 2023-06-23:
I have written a python script to do such, specifically for reprap. I changed this from one that was for marlin that i found on thingiverse. The link is included. This is a post processing script. 

`# Cura PostProcessingPlugin
# Author:   Dryw Filtiarn
# Date:     March 01, 2019

# Description:  This plugin will allow you to create calibration GCode for
#               testing various values for Jerk  on marlin-2.0.x
#               this plugin should be used alongside the attached test model
#               as can be found on Thingiverse https://www.thingiverse.com/thing:3463159

from ..Script import Script
from UM.Application import Application

class LinAdvanceCalibration(Script):
    def __init__(self):
        super().__init__()

    def getSettingDataString(self):
        return """{
            "name": "Pressure Advance Calibration 1.0",
            "key": "PAC1",
            "metadata": {},
            "version": 2,
            "settings":
            {
                "start":
                {
                    "label": "Starting value",
                    "description": "The value for Jerk  the print will start at. A value of 0.02 is the Marlin default value (range 0.01 - 0.10, default 0.02)",
                    "type": "float",
                    "default_value": 0.02,
                    "minimum_value": 0.01,
                    "maximum_value": 0.10,
                    "minimum_value_warning": 0.01,
                    "maximum_value_warning": 0.10
                },
                "stepsize":
                {
                    "label": "Stepping size",
                    "description": "The increment with which Jerk  will be increased every 20 layers (range 0.005 - 0.100, default 0.010)",
                    "type": "float",
                    "default_value": 0.010,
                    "minimum_value": 0.005,
                    "maximum_value": 0.100,
                    "minimum_value_warning": 0.005,
                    "maximum_value_warning": 0.100
                }, 
                "layeradjustedat":
                {
                    "label": "Layer Which Changes",
                    "description": "The increment with which Linear Davnce will be increased every x layers",
                    "type": "int",
                    "default_value": 20,
                    "minimum_value": 1,
                    "maximum_value": 100
                },
                "lcdfeedback":
                {
                    "label": "Display details on LCD?",
                    "description": "This setting will insert M117 gcode instructions, to display current Jerk  value is being used.",
                    "type": "bool",
                    "default_value": true
                }
            }
        }"""
    
    def execute(self, data):
        uselcd = self.getSettingValueByKey("lcdfeedback")
        start = self.getSettingValueByKey("start")
        incr = self.getSettingValueByKey("stepsize")
        # lps = 20
        lps = self.getSettingValueByKey("layeradjustedat")

        lcd_gcode = 'M118 S"M572 '
        jd_gcode = "M572 D0 S"

        jd = start
        i = 0

        for layer in data:
            lay_idx = data.index(layer)
            
            lcd_out = lcd_gcode + ('%.3f' % jd) + '"'
            jd_out = jd_gcode + ('%.3f' % jd)

            lines = layer.split("\n")
            for line in lines:
                if line.startswith(";LAYER:"):
                    lin_idx = lines.index(line)

                    if i % lps == 0:
                        lines.insert(lin_idx + 1, jd_out)
                        if uselcd:
                            lines.insert(lin_idx + 2, lcd_out)
                        jd += incr
                            
                    i += 1
            
            result = "\n".join(lines)
            data[lay_idx] = result

        return data

        # for layer in data:
        #     display_text = lcd_text + str(i)
        #     layer_index = data.index(layer)
        #     lines = layer.split("\n")
        #     for line in lines:
        #         if line.startswith(";LAYER:"):
        #             line_index = lines.index(line)
        #             lines.insert(line_index + 1, display_text)
        #             i += 1
        #     final_lines = "\n".join(lines)
        #     data[layer_index] = final_lines
            
        # return data
`