fieldOfView/Cura-LinearAdvanceSettingPlugin #31: Error 0 on Save/Export to Octo

thebelin, 2021-02-23:
This error is exposed on dual extruder systems when the linear advance settings don't match on the extruders.

It shows the error message with the text of "0" in the description in Cura
**Error text:**
```
2021-02-22 18:32:02,759 - ERROR - [MainThread] UM.Logger.logException [110]:   File "C:\Users\thebelin\AppData\Roaming\cura\4.8\plugins\LinearAdvanceSettingPlugin\LinearAdvanceSettingPlugin\LinearAdvanceSettingPlugin.py", line 191, in _filterGcode
2021-02-22 18:32:02,762 - ERROR - [MainThread] UM.Logger.logException [110]:     if not apply_factor_per_feature[extruder_nr]:
2021-02-22 18:32:02,764 - ERROR - [MainThread] UM.Logger.logException [110]: KeyError: 0
```

My immediate fix was achieved by removing the Linear Advance Setting Plugin.

fieldOfView, 2021-02-23:
This error would happen if per-feature linear advance were set for only some of the extruders. It should be fixed in a0ca056
Thanks for reporting.

I'll keep this issue open until a new version of the plugin is released to the Marketplace.

fieldOfView, 2021-02-23:
Also see #26

commit a0ca0569 (2021-02-23): Fix error when linear advance settings are not set for all extruders

Fixes #31
closed 2022-05-23 by fieldOfView