fieldOfView/Cura-LinearAdvanceSettingPlugin #32: Turn off Linear Advance for some printers

AndreasALu, 2021-03-07:
Hi!

I have several different printers configurated in Cura, some of them do not support Linear Advance, some do. If I install the plugin, it puts a M572 or M900 in every G-code file, which causes the printers that do not support LA to throw an error.

I tried to do a post-processing via Search&Replace, but it seems that the LA plugin runs after the Post Processing Plugin, therefor I cannot get rid of the unwanted command.

The Post Processing steps are saved per printer, so it would normally work.

Is there a way to deactivate the plugin per printer oder at least run the plugin before the Post Processing Plugin?

fieldOfView, 2021-03-07:
There currently is no way to deactivate the plugin for a certain printer. Thanks for the request, this is on my shortlist of things to fix for the next version of the plugin.

AndreasALu, 2021-03-07:
Thx for the quick reply. Would it be easier to run the LA plugin *before* the post processing plugin? That would solve my problem.

fieldOfView, 2021-03-07:
It would not solve mine ;-)

tutulino, 2021-03-30:
Same problem here, I have multiple printers and I usually use firmware configured Linear Advance, but some profiles I need different linear advance (for example 0.8 nozzle, or different materials) I think best way to solve all problems is add a check box at profile level to activate/deactivate linear advance control, same way Cura does for Jerk or acceleration.

This way you can install plugin and use for some printer or some profiles only

commit 9d951d60 (2021-07-06): Add global disable setting

Fixes #32