fieldOfView/Cura-LinearAdvanceSettingPlugin #33: No new line before M900

blazewicz, 2021-07-02:
### Context

Cura: v4.9.1
Plugin: v3.5.11

### Description

Below is fragment of generated G-code, just after the start G-code:

```gcode
G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:100M900 K0.750000 T0 ;added by LinearAdvanceSettingPlugin

;LAYER:0
M107
```

A newline is missing after `;LAYER_COUNT:100`.

fieldOfView, 2021-07-03:
Strange. Can you save a project (File -> Save project...) so I can see what combination of settings leads to this?

blazewicz, 2021-07-03:
Sure, I can't share it on GitHub so I emailed you.

blazewicz, 2021-07-03:
Sorry for false alert, it was my script adding M117 with current progress that mangled the output. Your plugin works fine when I disabled it.

closed 2021-07-03 by blazewicz