fieldOfView/Cura-LinearAdvanceSettingPlugin #34: Initial layer factor not showing up, if all the other factors are the same.

aLECChe, 2021-08-31:
Thank you very much for this great plugin. 

[image]

I use this plugin mostly to disable LA for the first layer. But it seems if I only set the Initial Layer Linear Advance Factor to 0. Nothing would happen.

[image]

There is only ONE line of M900 in the sliced G code file.

[image]

But if I changed any one of the other Factors in Cura, it seems the problem is gone.

[image]

There are multiple M900, with the M900 K0 just before the first layer.

I am wondering, is this a problem ? or did I miss something on the correct usage ? 

Thank you once again!

shyblower, 2022-02-22:
I also just stumbled over this issue

aLECChe, 2022-02-23:
my temporary solution is to use very similar value in other factors. something like:

wall factor: 0.55
top/ bottom: 0.54
initial layer: 0