fieldOfView/Cura-LinearAdvanceSettingPlugin #36: [BUG] (?) LinearAdvance Plugin: Abort prints in Cura / Marlin when LinearAdvance is activated (>0.0)

SpRiNgToIfEl, 2022-01-08:
**Application Version**
4.11.x and above

**Printer**
Ender 5 Plus with BTT SKR PRO 1.2 / BTT TFT50v3 / Marlin 2.9.0.1 and above

**Reproduction steps**
Installed Cura 4.11.x / 4.12.x / 4.13.0b
Installed LinearAdvance Plugin (latest version)
Sliced the Thingiverse Calibration Cube (Link Thingiverse) in Cura 4.11.x and above
Start Print: Ender 5 Plus with BTT SKR PRO 1.2 / BTT TFT50v3 / Marlin 2.9.0.1 and above.

**Actual results:**
In Marlin LinearAdvance ist activated (see attached files).
In Cura the PlugIn "LinearAdvance" is installed (see attached files).

_Test 1:_
Sliced in Cura 4.11.x and above without Linear Advance (0.0): Print is sucsesfull

_Test 2:_
Sliced in Cura 4.11.x and above with Linear Advance (>0.0): Print fails after 4-5 layers (Position of fail: random position in outer wall, inner wall or infill
[Configuration_2.9.0.1.zip](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/files/7833942/Configuration_2.9.0.1.zip)
[Configuration_2.9.0.3.zip](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/files/7833943/Configuration_2.9.0.3.zip)
[image]
[image]
)

SpRiNgToIfEl, 2022-01-09:
I am using TMC 2208v3 UART.

After some searching I found a "solution" to use LA on Marlin v2.9.0.3 (nightly):
- activate SQUARE_WAVE_STEPPING (beta)
- deactivate extruder STEALTHCHOP (extruder runs now with spreadcycle)

Now the e-stepper motor hums / beeps a bit (not annoying), but I can now print with LA.
If necessary, the TMC 2208 will changed in the future.

The background noise (extruder) should probably be reduced with higher steps on the extruder.

fieldOfView, 2022-01-10:
Unless there is a way for me to fix this with gcode, I am going to say this is a firmware/hardware issue and I can do nothing about that with this plugin.

closed 2022-01-10 by fieldOfView