fieldOfView/Cura-LinearAdvanceSettingPlugin #39: Documentation for Klipper users is incorrect

psiberfunk, 2022-04-03:
The documentation has an invalid GCODE M900 conversion macro that doesn't work on current versions of klipper (maybe past versions too?). Instead, you should update the readme to this valid alternative:

[gcode_macro M900]
gcode:
    # Parameters
    {% set pa = params.K|float %}
    SET_PRESSURE_ADVANCE ADVANCE={pa}

psiberfunk, 2022-04-03:
Note that when you use the current suggested GCODE macro in printer.cfg in a modern klipper setup, klipper refuses to start with an error.. it doesn't like the line about "default_parameter_K: 0".  I'm not sure if that syntax ever worked properly.

fieldOfView, 2022-05-20:
See https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/pull/37
I am awaiting confirmation if this fix depends on the version of Klipper being used.

drphil3d, 2022-05-26:
[gcode_macro M900]
description: Set Pressure Advance
gcode:
  SET_PRESSURE_ADVANCE ADVANCE={params.K|default(0)}

fieldOfView, 2023-04-24:
I think this has been fixed by #37

closed 2023-04-24 by fieldOfView