fieldOfView/Cura-LinearAdvanceSettingPlugin #40: Feature request: Checkbox to disable Linear Advance on-demand

Jilocasin, 2022-05-13:
Right now, using the plugin will always cause a M900 to be included in the GCODE, as far as I'm aware.

For some prints I don't want to override the firmware value. Leaving the Linear Advance factor at 0 will still include a M900 K0.

Can you include a toggle checkbox like "Enable Linear Advance Control", silimar to Curas "Enable Acceleration control"?

fieldOfView, 2022-05-15:
> Can you include a checkbox like "Enable Linear Advance Control", silimar to Curas "Enable Acceleration control" that will toggle the entire feature?

That is exactly what the next version of the plugin will have. It is currently awaiting approval for inclusion in the Marketplace.

Jilocasin, 2022-05-15:
Great, thank you!

fieldOfView, 2022-05-15:
I'll keep this open until the version that includes this is available on the Marketplace