fieldOfView/Cura-LinearAdvanceSettingPlugin #42: The plugin does not work properly with Cura 5.0.0

x40-Community, 2022-06-06:
Hello,
I installed Ultimaker Cura 5.0.0 and the Linear Advance Settings plugin version 3.6.0.

I entered the following into the material file:
       <cura:setting key="material_linear_advance_factor">1.43</cura:setting>

I entered the following into the intent profile:
material_linear_advance_enable = True

When I call up the corresponding profile in Cura (after Cura restart), Linear Advance is always deactivated. When activated, the values are visible.

Even if I put material_linear_advance_factor in the intent profile it doesn't work.

You always have to activate Linear Advance manually.
Bug or am I doing something wrong?

qwertymodo, 2022-08-25:
This bug still seems to be present in 5.1.0.  Also, the "Enable Linear Advance" option does not show up under the "Select settings" dialog, so there is no way to assign the enabled option to the material profile within the GUI.

qwertymodo, 2022-08-25:
Something interesting I noted looking into this which may be relevant, or may just be nothing, but it does work to save the enabled setting into a new quality profile, and when doing so, the key name is material_linear_advance_enable where most enable/disable settings use _enabled (with a 'd' on the end).

qwertymodo, 2022-08-26:
I found a workaround.  Inside the plugin install directory (on Windows it's %APPDATA%\cura\(version)\plugins\LinearAdvanceSettingPlugin\LinearAdvanceSettingPlugin), edit linear_advance_35.def.json and linear_advance47.def.json.  Near the top, under "material_linear_advance_enable", set "default_value" to true.  Because the default K-factor for materials you haven't set is 0, enabling LA by default with a default K-value of 0 is the same as disabling it, so this is an acceptable workaround for me until the plugin can be fixed properly.

nelsongraca, 2023-02-28:
@qwertymodo I do not see that option in v3.6.0, and yet it does not show up on cura 5.x

fieldOfView, 2023-03-20:
Sorry, it took me a while to figure out what the actual issue was. I think I have fixed this for the next release of the plugin. Could you help test this?
https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/releases/tag/v3.6.1-DEV
Download the .curapackage for your version of Cura, and drop it onto the buildplate in Cura as if you were opening a 3d model.

nelsongraca, 2023-03-20:
@fieldOfView Can confirm it's working, thank you.

closed 2023-04-24 by fieldOfView