fieldOfView/Cura-LinearAdvanceSettingPlugin #44: Culture aware plugin

GMagician, 2023-02-12:
Question 1:
  Is there any chance you may add languages to this plugin?
Question 2:
  If I try to realize what above in a PR (not sure I can do it), is there any chance it will be merged?

Thanks for your attention

fieldOfView, 2023-02-18:
Sorry I did not respond to this.
Yes, translations are welcome, and some of my other plugins already include some translations. However, the Linear Advance plugin is a little bit more complicated because the translations are mainly settings, as opposed to text in user interface elements (buttons etc).

GMagician, 2023-02-18:
No worry. I closed because I tryed to get documents about plugin in Cura but there are few documents, one should scan all cura code to see what can be used and what can't. Add ther fact I don't know phyton language and you've done it.

My original idea was to use builtin translation to change on the fly content you read for settings but I need to learn phyton a little more before even think of do it

Thanks for your reply

GMagician, 2023-02-18:
Sorry to bother. I can translate settings but what I miss is how to detect current language cura uses. Do you maybe know how to get it?

fieldOfView, 2023-03-05:
Translations shall now be made by creating a .po translation from the .pot template, using eg [poedit](https://poedit.net/). You do not have to commit the compiled .mo file; this is automatically created and kept up to date with a post-merge action. You need to use the master branch.

fieldOfView, 2023-03-13:
This seems to work now via https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/commit/45fc35b75217200b92b1a77e18af22b301598f58

GMagician, 2023-03-13:
Not sure I did it correctly but it doesn't work. I downloaded zip file and overwritten in my cura folder. Also added .mo (rebuilt from poedit) into it-it folder

GMagician, 2023-03-13:
It also creates a "script" folder inside plugin folder

fieldOfView, 2023-03-20:
Thanks for testing. I'll leave this issue open until the fixed version is available in the Marketplace.

closed 2023-04-24 by fieldOfView